lucasw/ros_from_src #11: compilation error from rosconsole, on ubuntu 22.04

youliangtan, 2022-08-31:
I noticed this repo while trying to install noetic on ubuntu 22.04. However, when first trying out this repo, Iam getting a compilation from `rosconsole`,error shown as below. This is tested on a fresh new 22.04 PC. 

```bash
Errors     << rosconsole:make /home/youliang/ros_ws/build/catkin_ws/logs/rosconsole/build.make.002.log                                               
/home/youliang/ros_ws/build/catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void ros::console::impl::initialize()’:
/home/youliang/ros_ws/build/catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:169:23: error: cannot convert ‘ros::console::impl::ROSConsoleStdioAppender*’ to ‘log4cxx::AppenderPtr’ {aka ‘std::shared_ptr<log4cxx::Appender>’}
  169 |   logger->addAppender(new ROSConsoleStdioAppender);
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                       |
      |                       ros::console::impl::ROSConsoleStdioAppender*
In file included from /usr/include/log4cxx/spi/loggingevent.h:28,
                 from /usr/include/log4cxx/layout.h:29,
                 from /usr/include/log4cxx/appenderskeleton.h:28,
                 from /home/youliang/ros_ws/build/catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:42:
/usr/include/log4cxx/logger.h:144:60: note:   initializing argument 1 of ‘virtual void log4cxx::Logger::addAppender(log4cxx::AppenderPtr)’
  144 |                 virtual void addAppender(const AppenderPtr newAppender);
      |                                          ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/home/youliang/ros_ws/build/catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void* ros::console::impl::getHandle(const string&)’:
/home/youliang/ros_ws/build/catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:203:36: error: cannot convert ‘log4cxx::LoggerPtr’ {aka ‘std::shared_ptr<log4cxx::Logger>’} to ‘void*’ in return
  203 |   return log4cxx::Logger::getLogger(name);
      |          ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
      |                                    |
```

youliangtan, 2022-09-01:
I manage to fix this compilation error, by replacing the `build/catkin_ws/src/rosconsole` pkg  with your forked version of `rosconsole`. 

https://github.com/lucasw/rosconsole , branch `concise_output_roso`. I think it will be nice if we can mention this error in the original `ros/rosconsole` repo, or even open a PR to upstream.

lucasw, 2022-09-01:
There is an issue and pr in rosconsole already: https://github.com/ros/rosconsole/issues/50

mgrallos, 2022-11-11:
Hi, new to this and I probably don't know what I'm doing 50% of the time. I encountered the same error but I tried replacing the ros/rosconsole folder with the forked version and command the build.sh again but it's still the same.

lucasw, 2022-11-12:
>  I tried replacing the ros/rosconsole folder with the forked version and command the build.sh again but it's still the same.

Was that after doing `vcs import --shallow < base_repos.yaml` ?  That base_repos.yaml already points to the concise_output_roso version.

https://github.com/lucasw/ros_from_src/blob/core/ubuntu_2204/base_repos.yaml#L440-L443

@mgrallos are you building on 22.04 and following this [22.04 readme](https://github.com/lucasw/ros_from_src/blob/core/ubuntu_2204/README.md)?