lucasw/ros_from_src #15: Problem with using this build

hiqsol, 2022-12-23:
Thanks to you I could build the ROS.

But, unfortunately, I failed to use it to build my ROS project.

What I did:
1. `source ros_from_src/build/ros/setup.bash`
2. `mkdir ~/catkin_ws` and download there my project
3. `cd ~/catkin_ws`

lucasw, 2022-12-23:
what 

Do you have libstd-msgs-dev, python3-std-msgs, & ros-std-msgs installed (through normal apt), `apt install ros-*` from the readme should have gotten them for you. 

Does `/usr/share/std_msgs/msg` exist on your system?  Or do you have any of these:

```
libstd-msgs-dev: /usr/share/pkgconfig/std_msgs.pc
libstd-msgs-dev: /usr/share/std_msgs/cmake/std_msgs-msg-extras.cmake
libstd-msgs-dev: /usr/share/std_msgs/cmake/std_msgs-msg-paths.cmake
libstd-msgs-dev: /usr/share/std_msgs/cmake/std_msgsConfig-version.cmake
libstd-msgs-dev: /usr/share/std_msgs/cmake/std_msgsConfig.cmake
```

hiqsol, 2022-12-28:
Yes I have those files installed.
But it is not enough.

What helped me was that I've sourced `ros_from_source/build/catkin_ws/devel/setup.bash`

closed 2023-01-05 by lucasw