lucasw/ros_from_src #16: How to install ROS built with this package

hiqsol, 2022-12-29:
Ok. Thanks to you, I have the ROS built on Ubuntu 22.04 and my project build with the ROS. :tada: :tada: :tada:

Now I need to spread it over multiple RPis.

Can you please advise me how to install the build?

Thank you!

lucasw, 2023-01-04:
You want to build it on one rpi, then copy the built files to the others to avoid the lengthy build?

Do a catkin install build, tar/zip up the `install/` directory and then copy and decompress on the target computers.  If you installed anything through apt install or pip or similar you'll need to repeat those steps because they won't be inside `install/`, if you don't you'll get exceptions or crashes when the built files try to use dependencies.

If you have custom ros packages you'll very likely have to modify their CMakeLists.txt to instruct the install build what files to install (like the contents of `scripts/` for example).

hiqsol, 2023-01-04:
Thank you!
I've tried to build-install ROS and then copy `build/catkin_ws/install` dir, but it turned to be not enough.
To build my ROS-based project I had to copy also `build/ros` directory.
Everything seems to work but now I'm not sure what is enough to copy to get the ROS built from source completely enough :)
If you know any manual on building installable ROS package please let know :) Any public repository doing so is also will help me very much cause I couldn't find how the ROS maintainers create their debian packages.