lucasw/ros_from_src #22: ROS Build fails for rosconsole

vidhiJain, 2023-02-21:
I tried your repo `ros_from_src` and the instructions on ros wiki `http://wiki.ros.org/noetic/Installation/Source` to install ROS Noetic on Ubuntu 22.04.
I see that the build process is failing to install `rosconsole`.
Running build commands `../ros_from_src/build.sh` for repo `ros_from_src`  or `./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release` as per ros wiki results in the same error.
Here is the error log:
```
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: regex system thread 
-- rosconsole backend: log4cxx
-- Configuring done
-- Generating done
-- Build files have been written to: /home/vidhij/ros_catkin_ws/build_isolated/rosconsole
==> make -j12 -l12 in '/home/vidhij/ros_catkin_ws/build_isolated/rosconsole'
[ 16%] Building CXX object CMakeFiles/rosconsole_backend_interface.dir/src/rosconsole/rosconsole_backend.cpp.o
[ 33%] Linking CXX shared library /home/vidhij/ros_catkin_ws/devel_isolated/rosconsole/lib/librosconsole_backend_interface.so
[ 33%] Built target rosconsole_backend_interface
[ 50%] Building CXX object CMakeFiles/rosconsole_log4cxx.dir/src/rosconsole/impl/rosconsole_log4cxx.cpp.o
/home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void ros::console::impl::initialize()’:
/home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:169:23: error: cannot convert ‘ros::console::impl::ROSConsoleStdioAppender*’ to ‘log4cxx::AppenderPtr’ {aka ‘std::shared_ptr<log4cxx::Appender>’}
  169 |   logger->addAppender(new ROSConsoleStdioAppender);
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                       |
      |                       ros::console::impl::ROSConsoleStdioAppender*
In file included from /usr/include/log4cxx/spi/loggingevent.h:28,
                 from /usr/include/log4cxx/layout.h:29,
                 from /usr/include/log4cxx/appenderskeleton.h:28,
                 from /home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:42:
/usr/include/log4cxx/logger.h:144:60: note:   initializing argument 1 of ‘virtual void log4cxx::Logger::addAppender(log4cxx::AppenderPtr)’
  144 |                 virtual void addAppender(const AppenderPtr newAppender);
      |                                          ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void* ros::console::impl::getHandle(const string&)’:
/home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:203:36: error: cannot convert ‘log4cxx::LoggerPtr’ {aka ‘std::shared_ptr<log4cxx::Logger>’} to ‘void*’ in return
  203 |   return log4cxx::Logger::getLogger(name);
      |          ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
      |                                    |
      |                                    log4cxx::LoggerPtr {aka std::shared_ptr<log4cxx::Logger>}
/home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘bool ros::console::impl::get_loggers(std::map<std::__cxx11::basic_string<char>, ros::console::levels::Level>&)’:
/home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:219:120: error: conversion from ‘log4cxx::spi::LoggerRepositoryWeakPtr’ {aka ‘std::weak_ptr<log4cxx::spi::LoggerRepository>’} to non-scalar type ‘log4cxx::spi::LoggerRepositoryPtr’ {aka ‘std::shared_ptr<log4cxx::spi::LoggerRepository>’} requested
  219 | :Logger::getLogger(ROSCONSOLE_ROOT_LOGGER_NAME)->getLoggerRepository();
      | ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~

/home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void ros::console::impl::register_appender(ros::console::LogAppender*)’:
/home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:361:23: error: cannot convert ‘ros::console::impl::Log4cxxAppender*’ to ‘log4cxx::AppenderPtr’ {aka ‘std::shared_ptr<log4cxx::Appender>’}
  361 |   logger->addAppender(g_log4cxx_appender);
      |                       ^~~~~~~~~~~~~~~~~~
      |                       |
      |                       ros::console::impl::Log4cxxAppender*
In file included from /usr/include/log4cxx/spi/loggingevent.h:28,
                 from /usr/include/log4cxx/layout.h:29,
                 from /usr/include/log4cxx/appenderskeleton.h:28,
                 from /home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:42:
/usr/include/log4cxx/logger.h:144:60: note:   initializing argument 1 of ‘virtual void log4cxx::Logger::addAppender(log4cxx::AppenderPtr)’
  144 |                 virtual void addAppender(const AppenderPtr newAppender);
      |                                          ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void ros::console::impl::deregister_appender(ros::console::LogAppender*)’:
/home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:368:27: error: no matching function for call to ‘log4cxx::Logger::removeAppender(ros::console::impl::Log4cxxAppender*&)’
  368 |     logger->removeAppender(g_log4cxx_appender);
      |     ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/log4cxx/spi/loggingevent.h:28,
                 from /usr/include/log4cxx/layout.h:29,
                 from /usr/include/log4cxx/appenderskeleton.h:28,
                 from /home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:42:
/usr/include/log4cxx/logger.h:1450:22: note: candidate: ‘virtual void log4cxx::Logger::removeAppender(log4cxx::AppenderPtr)’
 1450 |                 void removeAppender(const AppenderPtr appender);
      |                      ^~~~~~~~~~~~~~
/usr/include/log4cxx/logger.h:1450:55: note:   no known conversion for argument 1 from ‘ros::console::impl::Log4cxxAppender*’ to ‘log4cxx::AppenderPtr’ {aka ‘std::shared_ptr<log4cxx::Appender>’}
 1450 |                 void removeAppender(const AppenderPtr appender);
      |                                     ~~~~~~~~~~~~~~~~~~^~~~~~~~
/usr/include/log4cxx/logger.h:1456:22: note: candidate: ‘virtual void log4cxx::Logger::removeAppender(const LogString&)’
 1456 |                 void removeAppender(const LogString& name);
      |                      ^~~~~~~~~~~~~~
/usr/include/log4cxx/logger.h:1456:54: note:   no known conversion for argument 1 from ‘ros::console::impl::Log4cxxAppender*’ to ‘const LogString&’ {aka ‘const std::__cxx11::basic_string<char>&’}
[... 11 lines cut ...]
 1450 |                 void removeAppender(const AppenderPtr appender);
      |                      ^~~~~~~~~~~~~~
/usr/include/log4cxx/logger.h:1450:55: note:   no known conversion for argument 1 from ‘ros::console::impl::Log4cxxAppender*’ to ‘log4cxx::AppenderPtr’ {aka ‘std::shared_ptr<log4cxx::Appender>’}
 1450 |                 void removeAppender(const AppenderPtr appender);
      |                                     ~~~~~~~~~~~~~~~~~~^~~~~~~~
/usr/include/log4cxx/logger.h:1456:22: note: candidate: ‘virtual void log4cxx::Logger::removeAppender(const LogString&)’
 1456 |                 void removeAppender(const LogString& name);
      |                      ^~~~~~~~~~~~~~
/usr/include/log4cxx/logger.h:1456:54: note:   no known conversion for argument 1 from ‘ros::console::impl::Log4cxxAppender*’ to ‘const LogString&’ {aka ‘const std::__cxx11::basic_string<char>&’}
 1456 |                 void removeAppender(const LogString& name);
      |                                     ~~~~~~~~~~~~~~~~~^~~~
/home/vidhij/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:386:58: error: base operand of ‘->’ has non-pointer type ‘log4cxx::spi::LoggerRepositoryWeakPtr’ {aka ‘std::weak_ptr<log4cxx::spi::LoggerRepository>’}
  386 |   log4cxx::Logger::getRootLogger()->getLoggerRepository()->shutdown();
      |                                                          ^~
make[2]: *** [CMakeFiles/rosconsole_log4cxx.dir/build.make:76: CMakeFiles/rosconsole_log4cxx.dir/src/rosconsole/impl/rosconsole_log4cxx.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:368: CMakeFiles/rosconsole_log4cxx.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
<== Failed to process package 'rosconsole': 
  Command '['/home/vidhij/ros_catkin_ws/install_isolated/env.sh', 'make', '-j12', '-l12']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /home/vidhij/ros_catkin_ws/build_isolated/rosconsole && /home/vidhij/ros_catkin_ws/install_isolated/env.sh make -j12 -l12

Command failed, exiting.
```

berndpfrommer, 2023-05-11:
There is a workaround for this. When the build fails with the error message above, apply this fix to the workspace:
```
cd catkin_ws/src/rosconsole
git remote add logger_fix git@github.com:dreuter/rosconsole.git
git fetch logger_fix log4cxx-0.12
git merge logger_fix/log4cxx-0.12
```
Then finish the build
```
cd ../../../
../ros_from_src/build.sh
```

vignif, 2023-07-26:
Your solution worked greatly!
I installed it on a `ubuntu22.04.2 LTS` machine with `anaconda3` and `python3.10` and had to fix a `libuuid` error. 
Solution adapted from [this](https://github.com/uzh-rpg/rpg_esim/issues/7#issuecomment-687999099)
```
ls ~/anaconda3/lib/libuuid*
mkdir ~/anaconda3/libuuid
mv ~/anaconda3/lib/libuuid* ~/anaconda3/libuuid
```

rishikoushikz, 2024-10-28:
i also got the same error in fedora 40 `/usr/lib/python3.12/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
<== Finished processing package [61 of 184]: 'angles'

==> Processing catkin package: 'rosconsole'
==> Building with env: '/home/rishi/ros_catkin_ws/install_isolated/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/home/rishi/ros_catkin_ws/build_isolated/rosconsole'
==> make -j12 -l12 in '/home/rishi/ros_catkin_ws/build_isolated/rosconsole'
[ 33%] Built target rosconsole_backend_interface
[ 50%] Building CXX object CMakeFiles/rosconsole_log4cxx.dir/src/rosconsole/impl/rosconsole_log4cxx.cpp.o
/home/rishi/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void ros::console::impl::initialize()’:
/home/rishi/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:169:23: error: cannot convert ‘ros::console::impl::ROSConsoleStdioAppender*’ to ‘log4cxx::AppenderPtr’ {aka ‘std::shared_ptr<log4cxx::Appender>’}
  169 |   logger->addAppender(new ROSConsoleStdioAppender);
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                       |
      |                       ros::console::impl::ROSConsoleStdioAppender*
In file included from /usr/include/log4cxx/spi/loggingevent.h:23,
                 from /usr/include/log4cxx/layout.h:23,
                 from /usr/include/log4cxx/appenderskeleton.h:22,
                 from /home/rishi/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:42:
/usr/include/log4cxx/logger.h:85:52: note:   initializing argument 1 of ‘virtual void log4cxx::Logger::addAppender(log4cxx::AppenderPtr)’
   85 |                 void addAppender(const AppenderPtr newAppender) override;
      |                                  ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/home/rishi/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void ros::console::impl::print(void*, ros::console::Level, const char*, const char*, const char*, int)’:
/home/rishi/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:187:98: error: no matching function for call to ‘log4cxx::spi::LocationInfo::LocationInfo(const char*&, const char*&, int&)’
  187 |     logger->forcedLog(g_level_lookup[level], str, log4cxx::spi::LocationInfo(file, function, line));
      |                                                                                                  ^
In file included from /usr/include/log4cxx/logger.h:24:
/usr/include/log4cxx/spi/location/locationinfo.h:93:17: note: candidate: ‘log4cxx::spi::LocationInfo::LocationInfo(const log4cxx::spi::LocationInfo&)’
   93 |                 LocationInfo( const LocationInfo& src );
      |                 ^~~~~~~~~~~~
/usr/include/log4cxx/spi/location/locationinfo.h:93:17: note:   candidate expects 1 argument, 3 provided
/usr/include/log4cxx/spi/location/locationinfo.h:87:17: note: candidate: ‘log4cxx::spi::LocationInfo::LocationInfo()’
   87 |                 LocationInfo();
      |                 ^~~~~~~~~~~~
/usr/include/log4cxx/spi/location/locationinfo.h:87:17: note:   candidate expects 0 arguments, 3 provided
/usr/include/log4cxx/spi/location/locationinfo.h:79:17: note: candidate: ‘log4cxx::spi::LocationInfo::LocationInfo(const char*, const char*, const char*, int)’
   79 |                 LocationInfo( const char* const fileName,
      |                 ^~~~~~~~~~~~
/usr/include/log4cxx/spi/location/locationinfo.h:79:17: note:   candidate expects 4 arguments, 3 provided
/home/rishi/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘bool ros::console::impl::get_loggers(std::map<std::__cxx11::basic_string<char>, ros::console::levels::Level>&)’:
/home/rishi/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:224:120: error: conversion from ‘log4cxx::spi::LoggerRepository*’ to non-scalar type ‘log4cxx::spi::LoggerRepositoryPtr’ {aka ‘std::shared_ptr<log4cxx::spi::LoggerRepository>’} requested
  224 |   log4cxx::spi::LoggerRepositoryPtr repo = log4cxx::Logger::getLogger(ROSCONSOLE_ROOT_LOGGER_NAME)->getLoggerRepository();
      |                                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
/home/rishi/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp: In function ‘void ros::console::impl::deregister_appender(ros::console::LogAppender*)’:
/home/rishi/ros_catkin_ws/src/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:374:27: error: no matching function for call to ‘log4cxx::Logger::removeAppender(ros::console::impl::Log4cxxAppender*&)’
  374 |     logger->removeAppender(g_log4cxx_appender);
      |     ~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/usr/include/log4cxx/logger.h:1608:22: note: candidate: ‘virtual void log4cxx::Logger::removeAppender(log4cxx::AppenderPtr)’
 1608 |                 void removeAppender(const AppenderPtr appender) override;
[... cut ...]
make[1]: *** [CMakeFiles/Makefile2:318: CMakeFiles/rosconsole_log4cxx.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
<== Failed to process package 'rosconsole': 
  Command '['/home/rishi/ros_catkin_ws/install_isolated/env.sh', 'make', '-j12', '-l12']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /home/rishi/ros_catkin_ws/build_isolated/rosconsole && /home/rishi/ros_catkin_ws/install_isolated/env.sh make -j12 -l12

Command failed, exiting.
`  
can any one kindly help me to solve this issue