lucasw/ros_from_src #26: Ros distro showing "one"

mouhyemen, 2023-08-15:
I followed the README and completed all the steps until docker. On running `roscore`, I get the following:

[image]

Notice that the rosdistro shows "one"

On command line, I get the following output when running `rosversion ros`
```
mouhkhan@pop-os ~ $ rosversion ros
<unversioned>
```

I was expecting to see `noetic` as the distro. Did I miss anything?

lucasw, 2023-08-28:
My roscore reports `/rosdistro: Debian`, same with `rosversion --distro`- but I'm using the pre-built debian packages (following https://github.com/lucasw/ros_from_src/blob/robot_state_publisher/ubuntu_2204/README.md), not building completely from source.

I'm not clear where that version string comes from, rosversion is in https://github.com/ros-infrastructure/rospkg, but maybe ttps://github.com/ros-infrastructure/rosdistro or does that interact with files from somewhere else to determine version?

`one` I think is the same as 'ros-o' so your system reporting that is accurate, and ros-o and the debianized versions are mostly the same- what is the issue with it not reporting as `noetic`?

lucasw, 2023-08-28:
I finally tracked it down to this code which for me is in `/usr/lib/python3/dist-packages/rospkg/rosversion.py`:
```
    if args.distro:
        if 'ROS_DISTRO' in os.environ:
            distro_name = os.environ['ROS_DISTRO']
        else:
            distro_name = get_distro_name_from_roscore()
        if not distro_name:
            distro_name = 'Debian'
        printer(distro_name)
        sys.exit(0)
```

In the noetic version it would have been set to unknown https://github.com/ros-infrastructure/rospkg/blob/master/src/rospkg/rosversion.py#L118C1-L126C20

```
        if not distro_name:
            distro_name = '<unknown>'
```

Which means your distro is coming from `get_distro_name_from_roscore()` or `$ROS_DISTRO`