lucasw/ros_from_src #29: ros2 from source on Ubuntu 23.04

lucasw, 2023-09-25:


lucasw, 2023-09-25:
```
mkdir -p ~/ros/ros2_iron/src
cd ~/ros/ros2_iron/src
wget https://raw.githubusercontent.com/ros2/ros2/iron/ros2.repos
vcs import < ros2.repos
```

lucasw, 2023-09-25:
`--parallel-workers 4` doesn't mean limit to 4 cpu cores, rclpy looks to be taking many cores

lucasw, 2023-09-25:
Ran into issue building rviz_rendering where it wanted the xaw library, so installed libxaw7-dev and ran colcon build and it failed the same way, had to delete build/rviz_rendering and build again then it worked- it didn't search for the library again, was using cached results.

```
/usr/bin/ld: cannot find -lXAW_LIBRARY-NOTFOUND: No such file or directory
```

lucasw, 2023-09-26:
# ros1_bridge

It doesn't like the debian installed ros1 opencv_apps

```
CMake Error at cmake/find_ros1_package.cmake:29 (message):
  find_ros1_package() failed to find 'opencv_apps' using pkg_check_modules()
Call Stack (most recent call first):
  CMakeLists.txt:60 (find_ros1_package)
```

```
rospack find opencv_apps
/usr/share/opencv_apps
```

lucasw, 2023-09-26:
Can skip not-found packages in ros1_bridge:

```
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -57,8 +57,14 @@ foreach(ros1_message_package ${ros1_message_packages})
   # TODO(karsten1987): This is currently a workaround to work with ROS 2 classloader
   # rather than ROS 1 classloader.
   if(NOT "${ros1_message_package}" STREQUAL "nodelet")
-    find_ros1_package(${ros1_message_package} REQUIRED)
-    list(APPEND prefixed_ros1_message_packages "ros1_${ros1_message_package}")
+    find_ros1_package(${ros1_message_package})  # REQUIRED)
+    message(INFO " msg pkg: ${ros1_message_package} ${ros1_${ros1_message_package}_FOUND}")
+
+    if(NOT ros1_${ros1_message_package}_FOUND)
+      message(WARNING " skipping ${ros1_message_package}")
+    else()
+      list(APPEND prefixed_ros1_message_packages "ros1_${ros1_message_package}")
+    endif()
   endif()
 endforeach()
```

But then end up with

```
CMake Warning at CMakeLists.txt:197 (add_executable):
  Cannot generate a safe runtime search path for target dynamic_bridge
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libmessage_filters.so.1d] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/lucasw/install_base_catkin_ws/install/lib
    runtime library [libroscpp.so.4d] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/lucasw/install_base_catkin_ws/install/lib
    runtime library [libxmlrpcpp.so.3d] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/lucasw/install_base_catkin_ws/install/lib
    runtime library [librosconsole.so.3d] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/lucasw/install_base_catkin_ws/install/lib
    runtime library [librosconsole_log4cxx.so.3d] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/lucasw/install_base_catkin_ws/install/lib
    runtime library [librosconsole_backend_interface.so.3d] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/lucasw/install_base_catkin_ws/install/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  CMakeLists.txt:259 (custom_executable)

```

and error out with

```
In file included from /home/lucasw/ros/ros2_iron_overlay/build/ros1_bridge/generated/get_factory.cpp:32:
/home/lucasw/ros/ros2_iron_overlay/build/ros1_bridge/generated/tf2_msgs_factories.hpp:14:10: fatal error: tf2_msgs/msg/tf2_error.hpp: No such file or directory
   14 | #include <tf2_msgs/msg/tf2_error.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

lucasw, 2023-09-26:
Trying again clean while only sourcing `install_base_catkin_ws/devel/setup.bash` then `source ros2_iron/install/local_setup.bash` - went for ten minutes but appears to have worked with only warnings (same as above), skipping these:

```
 apriltag_ros
 base_local_planner
 costmap_2d 
 find_object_2d 
 frei0r_image 
 fuse_models 
 image_view2 
 jsk_rviz_plugins 
 jsk_topic_tools 
 laser_assembler 
 navfn 
 nodelet_rosbag 
 opencv_apps 
 polled_camera 
 robot_localization 
 rviz
```

lucasw, 2023-09-26:
```
~/ros/ros2_iron_overlay$ ros2 run ros1_bridge dynamic_bridge
failed to create 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log': No template specialization for the pair
check the list of supported pairs with the `--print-pairs` option
```

lucasw, 2023-09-26:
The example works, have to specify the topic type if there's no other subscriber, the topics aren't visible to ros2 topic list need to know what they are from the ros1 side:

```
rosrun rospy_tutorials talker
```

```
ros2 topic echo /chatter std_msgs/msg/String
```

lucasw, 2023-09-27:
Need rqt_image_view

Using rolling branch:

```
CMake Error at CMakeLists.txt:72 (target_link_libraries):
  Target "rqt_image_view" links to:

    cv_bridge::cv_bridge

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

```

lucasw, 2023-09-27:
Going to uninstall for now, need to check if install_base_catkin_ws still builds after this:
```
sudo apt remove libsensor-msgs-dev
```

lucasw, 2023-09-27:
Now can build rqt_image_view

```
In file included from /home/lucasw/ros/ros2_iron/src/misc/rqt_image_view/src/rqt_image_view/image_view.cpp:38:
/home/lucasw/ros/ros2_iron/install/cv_bridge/include/cv_bridge/cv_bridge/cv_bridge.h:42:2: warning: #warning This header is obsolete, please include cv_bridge/cv_bridge.hpp instead [-Wcpp]
   42 | #warning This header is obsolete, please include cv_bridge/cv_bridge.hpp instead
      |  ^~~~~~~
---
```

lucasw, 2023-09-27:
Try out rviz

```
ros2 run rviz2 rviz2
```

lucasw, 2023-09-27:
rclpy Duration doesn't have a `to_sec()`, the equivalent is `my_duration.nanoseconds / 1e9`

lucasw, 2023-09-28:
After full build didn't see an issue with `/usr/include/sensor_msgs/image_encodings.h` on a different system than the above- so didn't need to remove libsensor-msgs-dev

lucasw, 2023-09-28:
Need to add `--cmake-args -DCMAKE_BUILD_TYPE=Release` to colcon build https://github.com/colcon/colcon-cmake/issues/96

kennyS99, 2023-10-07:
> ```
> ~/ros/ros2_iron_overlay$ ros2 run ros1_bridge dynamic_bridge
> failed to create 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log': No template specialization for the pair
> check the list of supported pairs with the `--print-pairs` option
> ```
> 
> -> [ros2/ros1_bridge#391](https://github.com/ros2/ros1_bridge/issues/391)

Hey Lucas,
This issue still exists in the ros2 rolling version. Is there any solution you know to solve it? 
I appreciate it your replay and attention.

lucasw, 2023-10-07:
> Hey Lucas, This issue still exists in the ros2 rolling version. Is there any solution you know to solve it? I appreciate it your replay and attention.

I didn't find a solution, just ignored the error- maybe https://github.com/ros2/ros1_bridge/issues/391#issuecomment-1472665701 is a good start- is it possible to go back to https://github.com/ros2/ros1_bridge/commit/81f8b08671c8d161a0f0482bae4580d4d01e69c9 or are following commits necessary for it to work?

kennyS99, 2023-10-08:
> > Hey Lucas, This issue still exists in the ros2 rolling version. Is there any solution you know to solve it? I appreciate it your replay and attention.
> 
> I didn't find a solution, just ignored the error- maybe [ros2/ros1_bridge#391 (comment)](https://github.com/ros2/ros1_bridge/issues/391#issuecomment-1472665701) is a good start- is it possible to go back to [ros2/ros1_bridge@81f8b08](https://github.com/ros2/ros1_bridge/commit/81f8b08671c8d161a0f0482bae4580d4d01e69c9) or are following commits necessary for it to work?

Firstly, I very appreciate your quick replay. 
after a couple of attempts of commit 81f8b08671c8d161a0f0482bae4580d4d01e69c9, I found it could not solve my problem. but commit b9f1739fd84fc877a8ec6e5c416b65aa2d782f89 did solve the /rosout topics bridge very well.

lucasw, 2023-10-08:
Trying out [ros2/ros1_bridge@81f8b08](https://github.com/ros2/ros1_bridge/commit/81f8b08671c8d161a0f0482bae4580d4d01e69c9)

```
roscore
rostopic pub /rosout rosgraph_msgs/Log "{level: 0, name: 'foo', msg: 'bar123'}" -r 1
...
```

```
ros2 run ros1_bridge dynamic_bridge
created 1to2 bridge for topic '/rosout' with ROS 1 type 'rosgraph_msgs/Log' and ROS 2 type 'rcl_interfaces/msg/Log'
created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log'
[WARN] [1696773603.395194390] [ros_bridge]: New subscription discovered on topic '/rosout', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: DURABILITY_QOS_POLICY
[INFO] [1696773603.407521100] [ros_bridge]: Passing message from ROS 2 rcl_interfaces/msg/Log to ROS 1 rosgraph_msgs/Log (showing msg only once per type)
[INFO] [1696773604.067792874] [ros_bridge]: Passing message from ROS 1 rosgraph_msgs/Log to ROS 2 rcl_interfaces/msg/Log (showing msg only once per type)
```

```
ros2 topic echo /rosout
[WARN] [1696773674.406063393] [_ros2cli_103516]: New publisher discovered on topic '/rosout', offering incompatible QoS. No messages will be received from it. Last incompatible policy: DURABILITY
stamp:
  sec: 1696773674
  nanosec: 393526055
level: 30
name: ros_bridge
msg: New subscription discovered on topic '/rosout', requesting incompatible QoS. No messages will be sent to it. Last incompatible p...
file: /home/lucasw/ros/ros2_iron/src/ros2/rclcpp/rclcpp/src/rclcpp/publisher_base.cpp
function: default_incompatible_qos_callback
line: 332
---
```

Trying https://github.com/ros2/ros1_bridge/commit/b9f1739fd84fc877a8ec6e5c416b65aa2d782f89 next:

```
```

kennyS99, 2023-10-08:
> Trying out [ros2/ros1_bridge@81f8b08](https://github.com/ros2/ros1_bridge/commit/81f8b08671c8d161a0f0482bae4580d4d01e69c9) - it builds okay without the need to skip packages
> 
> ```
> roscore
> rostopic pub /rosout rosgraph_msgs/Log "{level: 0, name: 'foo', msg: 'bar123'}" -r 1
> ...
> ```
> 
> ```
> ros2 run ros1_bridge dynamic_bridge
> created 1to2 bridge for topic '/rosout' with ROS 1 type 'rosgraph_msgs/Log' and ROS 2 type 'rcl_interfaces/msg/Log'
> created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log'
> [WARN] [1696773603.395194390] [ros_bridge]: New subscription discovered on topic '/rosout', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: DURABILITY_QOS_POLICY
> [INFO] [1696773603.407521100] [ros_bridge]: Passing message from ROS 2 rcl_interfaces/msg/Log to ROS 1 rosgraph_msgs/Log (showing msg only once per type)
> [INFO] [1696773604.067792874] [ros_bridge]: Passing message from ROS 1 rosgraph_msgs/Log to ROS 2 rcl_interfaces/msg/Log (showing msg only once per type)
> ```
> 
> ```
> ros2 topic echo /rosout
> [WARN] [1696773674.406063393] [_ros2cli_103516]: New publisher discovered on topic '/rosout', offering incompatible QoS. No messages will be received from it. Last incompatible policy: DURABILITY
> stamp:
>   sec: 1696773674
>   nanosec: 393526055
> level: 30
> name: ros_bridge
> msg: New subscription discovered on topic '/rosout', requesting incompatible QoS. No messages will be sent to it. Last incompatible p...
> file: /home/lucasw/ros/ros2_iron/src/ros2/rclcpp/rclcpp/src/rclcpp/publisher_base.cpp
> function: default_incompatible_qos_callback
> line: 332
> ---
> ```
> 
> Trying [ros2/ros1_bridge@b9f1739](https://github.com/ros2/ros1_bridge/commit/b9f1739fd84fc877a8ec6e5c416b65aa2d782f89) next (but ros1_bridge takes a long time to compile...):

Thx for your reply.
Yes, I used this commit today, and it's working well with my env.
Best regards

lucasw, 2024-01-10:
```
sudo apt install liblttng-ctl-dev```

ros2_tracing

closed 2024-03-10 by lucasw