lucasw/ros_from_src #36: Build moveit from source in 22.04/23.04

lucasw, 2024-01-14:
https://github.com/ros-planning/moveit

lucasw, 2024-01-14:
```
catkin build -j1 moveit_core
...
CMake Error at /usr/lib/cmake/pybind11/pybind11NewTools.cmake:189 (python3_add_library):
  Unknown CMake command "python3_add_library".
```

lucasw, 2024-01-14:
```
Errors     << moveit_core:make /home/lucasw/ros/ros1_rse/logs/moveit_core/build.make.001.log                                                                                                                                        
In file included from /home/lucasw/ros/ros1_rse/src/moveit_all/moveit/moveit_core/trajectory_processing/src/ruckig_traj_smoothing.cpp:40:
/home/lucasw/ros/ros1_rse/src/moveit_all/moveit/moveit_core/trajectory_processing/include/moveit/trajectory_processing/ruckig_traj_smoothing.h:42:10: fatal error: ruckig/ruckig.hpp: No such file or directory
   42 | #include <ruckig/ruckig.hpp>
      |          ^~~~~~~~~~~~~~~~~~~
```

lucasw, 2024-01-14:
nlohmann-json3-dev -> json/json.hpp

lucasw, 2024-01-14:
```
sudo apt install libompl-dev
```

lucasw, 2024-01-14:
```
pip install eigenpy
```

lucasw, 2024-01-14:
```
repositories:
  eigenpy:
    type: git
    url: https://github.com/stack-of-tasks/eigenpy
    version: master
  geometric_shapes:
    type: git
    url: https://github.com/ros-planning/geometric_shapes.git
    version: noetic-devel
  graph_msgs:
    type: git
    url: git@github.com:PickNikRobotics/graph_msgs.git
    version: jade-devel
  moveit:
    type: git
    url: git@github.com:lucasw/moveit.git
    version: pybind_find_order
  moveit_msgs:
    type: git
    url: https://github.com/ros-planning/moveit_msgs.git
    version: master
  moveit_resources:
    type: git
    url: https://github.com/ros-planning/moveit_resources.git
    version: master
  moveit_tutorials:
    type: git
    url: https://github.com/ros-planning/moveit_tutorials.git
    version: master
  moveit_visual_tools:
    type: git
    url: https://github.com/ros-planning/moveit_visual_tools.git
    version: master
  object_recognition_msgs:
    type: git
    url: git@github.com:wg-perception/object_recognition_msgs.git
    version: master
  pybind11_catkin:
    type: git
    url: git@github.com:wxmerkt/pybind11_catkin.git
    version: master
  ruckig:
    type: git
    url: git@github.com:pantor/ruckig.git
    version: main
  rviz_visual_tools:
    type: git
    url: https://github.com/PickNikRobotics/rviz_visual_tools
    version: master
  srdfdom:
    type: git
    url: https://github.com/ros-planning/srdfdom
    version: noetic-devel
  warehouse_ros:
    type: git
    url: git@github.com:ros-planning/warehouse_ros.git
    version: kinetic-devel
```

lucasw, 2024-01-19:
@v4hn what about the third_party/json headers- I had to do this https://github.com/lucasw/ruckig/commit/2edf595617687e847e0b30d8776020a498a33d3e

v4hn, 2024-01-19:
I always build without the cloud client because I don't see the point in having json rpc interfaces in a library that usually runs on realtime systems on controllers.
That dependency was also the reason why the noetic package was(/is?) broken.